MillenniumMachines/millenniummachines.github.io #25: fastener count on the FMJ document is wrong (is probably 1/2 needed)

joshmurrah, 2024-03-06:
**Describe the bug**
The FMJ document has a BOM with number of fasteners - the number of fasteners is really short, it's probably the count for one plate and not both.

**Document Name / Location**
(https://millenniummachines.github.io/docs/milo/assembly_manual/chapters/fmj_assembly/#fmj-installation)

**Required Resolution**
adjust fastener count - I show 34 shown in the document photo, so BOM should be 68 - but adjust as needed.

**Screenshots**
[image]

EdwardChamberlain, 2024-03-06:
Thanks - corrected to 68 (34 per side) as per image. 

This will show up when the dev docs merge to main.

dmason1992, 2024-03-06:
#23 fixes this

commit 277a2a43 (2024-03-06): correct bolt qty fixes #25
closed 2024-03-06 by dmason1992