MillenniumMachines/millenniummachines.github.io #86: Wrong size for Y Axis Leadscrew, Tr8x8, 535mm

schoenid, 2024-09-08:
**Describe the bug**
In the documentation https://millenniummachines.github.io/docs/milo/manual/chapters/40_y_axis_assembly/#uninstall-the-y-axis-bearing-block the Y Axis Leadscrew is documented with 550 mm.
In the package I received, there is a Y Axis Leadscrew, Tr8x8, 535mm.
See also https://docs.ldomotors.com/en/milo/milov15/BOM_RevA

**Document Name / Location**
The location that has the issue:

https://millenniummachines.github.io/docs/milo/manual/chapters/40_y_axis_assembly/#uninstall-the-y-axis-bearing-block
https://millenniummachines.github.io/docs/milo/bom/sourcing_guide/#motion

**Required Resolution**
Please change the length in the description to 535mm.
Change the sourcing guide to: 
TR8x8 4-Start Lead-screw 535mm  1
TR8x8 4-Start Lead-screw 550mm  1

**Additional context**
If started with assembling the Y Axis like suggested, the wrong size of the Leadscrew will be used and the whole thing has to be disassembled and corrected later, while realizing, that the X Axis Leadscrew is to short, because only the one with 535mm is left.

schoenid, 2024-09-08:
Actually in the CAD file the length of `TR8x8 550mm` is 529 mm ...

Model tree in CAD.step:
```
Millenium Milo V1.5 (Release Version 1.6) v1
Minimill V2
Y Axis
TR8x8 550mm
```

jason7594, 2025-01-23:
Thanks for  point this, and LDO will adjust the document after Chinese New Year. Thanks!

benvanik, 2025-02-01:
I just ran into this during assembly - for those looking, more clearly: the Y lead screw is 535, the X lead screw is 550.

[image]

closed 2026-11-08 by avnc